lewisxhe/esp32-camera-series #2: Camera init fail

cc13com, 2019-02-08:
Hello,

Me again. :-)

I installed Arduino 1.8.8.

Added `https://github.com/espressif/arduino-esp32/releases/download/1.0.1-rc2/package_esp32_dev_index.json` to the board URL and `esp8266-oled-ssd1306-master.zip`, `OneButton-master.zip` to the libraries. In `Adafruit_BME280.h` I comment row 29 out.

As board I set `TTGO LoRa32-OLED V1`. Compiling and transfer to the board works fine. After that the display is showing "Camera init fail".

Could you help me please?

lewisxhe, 2019-02-09:
> Hello,
>
> Me again. :-)
>
> I installed Arduino 1.8.8.
>
> Added `https://github.com/espressif/arduino-esp32/releases/download/1.0.1-rc2/package_esp32_dev_index.json` to the board URL and [...]

Please change the board to T-Beam and make sure the PSAM is in ENABLE

cc13com, 2019-02-11:
Thanks, it works fine for now.

closed 2019-02-11 by cc13com