lewisxhe/esp32-camera-series #3: camera display?

richipower, 2019-02-08:
hello, i am new on the field and i just bought the camera to test it...
I connect the camera with my phone over the wifi but i dont know if i need an additional app for my cellphone? to display the data while connected with the camera?
do i need an extra arduino? just for the visualization?

thanks and greets

richipower, 2019-02-08:
ah sorry i got it working but in my phone i dont have the last two options:
- Face Detection
- Face Recognition
do i need to do something else?

lewisxhe, 2019-02-09:
> ah sorry i got it working but in my phone i dont have the last two options:
> 
> * Face Detection
> * Face Recognition
> 
> After color bar comes the camera stream... do i need to do something else?

The reason I hide the bottom button is because I am afraid of being misunderstood because the face recognition feature is not included in the code. If you need to run face recognition, run [esp-who](Https://github.com/espressif/esp-who)

closed 2019-02-14 by lewisxhe