lewisxhe/esp32-camera-series #4: Compiler error:  undefined reference to `startCameraServer()'         [ Help needed, please ]

Schoose, 2019-02-08:
Hi,
i did all the points, described in the wiki: Installation of the libraries and commented out the sensor.h in Adafruit_BME280.h. Unfortunatelly i get the following error while compiling:

sketch/esp32-camera-bme280.ino.cpp.o:(.literal._Z5setupv+0x6c): **undefined reference to `startCameraServer()'**

sketch/esp32-camera-bme280.ino.cpp.o: In function `setup()':
/home/dragon/Arduino/Sketchbook/esp32-camera-bme280-master/esp32-camera-bme280/esp32-camera-bme280.ino:291: undefined reference to `startCameraServer()'
collect2: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board ESP32 Wrover Module.

I tried to compile with the" ESP32 Wrover Module" and with "TTGO LoRa32-OLED V1" -Boards and got always the same error.
Could you help me please. Thanks a lot

lewisxhe, 2019-02-09:
> Hi,
> i did all the points, described in the wiki: Installation of the libraries and commented out the sensor.h in Adafruit_BME280.h. Unfortunatelly i get the following error while compiling: [...]

This is the project directory was reset by Arduino, please open the project directory and rename `esp32-camera-bme280-master` to `esp32-camera-bme280`, and ensure that the contents are the same as the following figure.
[image]

Schoose, 2019-02-09:
Thank you very much, now it works perfect, with " ESP32 Wrover Module" as Board.

closed 2019-02-09 by Schoose

Slavdos, 2019-05-08:
> > Привет, 
> > я сделал все пункты, описанные в вики: Установка библиотек и закомментировал sensor.h в Adafruit_BME280.h. К сожалению, я получаю следующую ошибку при компиляции:
> > sketch / esp32-camera-bme280.ino.cpp.o :(. literal._Z5setupv + 0x6c): **неопределенная ссылка на startCameraServer ()**
> > sketch / esp32-camera-bme280.ino.cpp.o: В функции `setup()': /home/dragon/Arduino/Sketchbook/esp32-camera-bme280-master/esp32-camera-bme280/esp32-camera-bme280.ino:291: undefined reference to `startCameraServer () ' 
> > collect2: error: ld возвращено 1 состояние 
> > выхода из состояния выхода 1 
> > Fehler beim Kompilieren für das Плата ESP32 Wrover Module.
> > Я попытался скомпилировать с "ESP32 Wrover Module" и с "TTGO LoRa32-OLED V1" -Boards и получил всегда одну и ту же ошибку. 
> > Не могли бы вы мне помочь, пожалуйста. большое спасибо
> 
> Этот каталог проекта был сброшен Arduino, пожалуйста, откройте каталог проекта и переименуйте `esp32-camera-bme280-master`его `esp32-camera-bme280`, и убедитесь, что содержимое совпадает со следующим рисунком.
> [image]

Slavdos, 2019-05-08:
Good day friends!
I ask for help, since I have a similar problem on Windows 10, arduino ide and ESP 32 ttgo T-camera. I tried to follow the examples of the solution that was described above, but nothing changes and I still could not compile the code.
I am new to this and ask you, please help me to deal with this problem. Thank you very much in advance.

Slavdos, 2019-05-08:
#include <WiFi.h>
#include <OneButton.h>
#include "freertos/event_groups.h"
#include <Wire.h>
#include <Adafruit_BME280.h>
#include "esp_camera.h"
#include "esp_wifi.h"
/***************************************
 *  Board select
 **************************************/
//! If you use OV2640_V16 microphone version, please enable this macro
// #define TTGO_OV2640_V16 

/***************************************
 *  Modules
 **************************************/
#define ENABLE_SSD1306
// #define SOFTAP_MODE       //The comment will be connected to the specified ssid
// #define ENABLE_BME280
#define ENABLE_SLEEP
#define ENABLE_IP5306

/***************************************
 *  WiFi
 **************************************/
#define WIFI_SSID   "Tenda_3A11A0"
#define WIFI_PASSWD "38120795"

/***************************************
 *  PinOUT
 **************************************/
#ifdef TTGO_OV2640_V16
#define PWDN_GPIO_NUM       -1
#define RESET_GPIO_NUM      -1
#define XCLK_GPIO_NUM       4
#define SIOD_GPIO_NUM       18
#define SIOC_GPIO_NUM       23

#define Y9_GPIO_NUM         36
#define Y8_GPIO_NUM         15
#define Y7_GPIO_NUM         12
#define Y6_GPIO_NUM         39
#define Y5_GPIO_NUM         35
#define Y4_GPIO_NUM         14
#define Y3_GPIO_NUM         13
#define Y2_GPIO_NUM         34
#define VSYNC_GPIO_NUM      5
#define HREF_GPIO_NUM       27
#define PCLK_GPIO_NUM       25
#define AS312_PIN           19
#define BUTTON_1            0
#undef ENABLE_BME280
#undef ENABLE_IP5306
#else
#define PWDN_GPIO_NUM       26
#define RESET_GPIO_NUM      -1
#define XCLK_GPIO_NUM       32
#define SIOD_GPIO_NUM       13
#define SIOC_GPIO_NUM       12

#define Y9_GPIO_NUM         39
#define Y8_GPIO_NUM         36
#define Y7_GPIO_NUM         23
#define Y6_GPIO_NUM         18
#define Y5_GPIO_NUM         15
#define Y4_GPIO_NUM         4
#define Y3_GPIO_NUM         14
#define Y2_GPIO_NUM         5
#define VSYNC_GPIO_NUM      27
#define HREF_GPIO_NUM       25
[... 271 lines cut ...]
#endif

#ifdef SOFTAP_MODE
    ip = WiFi.softAPIP().toString();
    Serial.printf("\nAp Started .. Please Connect %s hotspot\n", buff);
#else
    ip = WiFi.localIP().toString();
#endif

    Serial.print("Camera Ready! Use 'http://");
    Serial.print(ip);
    Serial.println("' to connect");
}

void loop()
{
#ifdef ENABLE_SSD1306
    if (ui.update()) {
#endif
        button1.tick();
#ifdef ENABLE_SSD1306
    }
#endif
}
Here is the code I use

mgnthe54, 2020-08-20:
Hello, I am having the same problem as on this thread: 
 
https://github.com/lewisxhe/esp32-camera-series/issues/4
 
I cannot locate the file: esp32-camera-bme280-master
 
how to access the picture you are showing? :
[image]
here are my error messages :
Arduino: 1.8.13 (Windows 7), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"
 
sketch\sketch_aug11b.ino.cpp.o:(.literal._Z5setupv+0x38): undefined reference to `startCameraServer()'
 
sketch\sketch_aug11b.ino.cpp.o: In function `setup()':
 
C:\Users\PC bureau Gilles\Documents\Arduino\sketch_aug11b/sketch_aug11b.ino:136: undefined reference to `startCameraServer()'
 
collect2.exe: error: ld returned 1 exit status
 
Multiple libraries were found for "WiFi.h"
 
Used: C:\Users\PC bureau Gilles\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 
Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
 
exit status 1
 
Error compiling for board ESP32 Dev Module.
 
 
Thank you for all help!
 
 
This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

paynterf, 2025-01-29:
I had this same problem recently with the 'CameraWebServer' example project (C:\Users\Frank\Documents\Arduino\Libraries\arduino-esp32-master\libraries\ESP32\examples\Camera\CameraWebServer).  The problem arises if the 'app_httpd.cpp' source file that contains the definition for 'startCameraServer()' and 'setupLedFlash(int pin)' isn't in the compile/link set for the project.  I use MS Visual Studio 2022 with Visual Micro for my Arduino projects, so for me the solution was to simply add app_httpd.cpp to the project (right click on project name in Solution Explorer ->Add -> Existing Item -> app_httpd.cpp).  After doing that everything compiled and ran fine