lewisxhe/esp32-camera-series #5: TTGO-Camera

collote69, 2019-02-14:
Hello sorry, I have the same error but I comment what you say and I use ESP32 Wrover Module but it tells me an error of compiling a card, I do not know what to do thanks.

lewisxhe, 2019-02-15:
> Hello sorry, I have the same error but I comment what you say and I use ESP32 Wrover Module but it tells me an error of compiling a card, I do not know what to do thanks.

Please submit an error message

collote69, 2019-02-15:
esp32-camera-bme280:5:29: error: Adafruit_Sensor.h: No such file or directory

Several libraries were found for "WiFi.h"
compilation terminated.

used: C:\Users\home\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\WiFi
 not used: C:\Users\home\Pictures\esp32\arduino-1.8.8\libraries\WiFi
exit status 1
Adafruit_Sensor.h: No such file or directory
and installed the sensor libraries but I do not know if I do it right I'm a bit of a rookie with this

lewisxhe, 2019-02-15:
> esp32-camera-bme280:5:29: error: Adafruit_Sensor.h: No such file or directory
>
> Several libraries were found for "WiFi.h"
> compilation terminated.
>
> used: [...]

In order to support BME280, I will use Adafruit_BME280_Library, but this library conflicts with <esp_camera.h> and enters Adafruit_BME280_Library In Adafruit_BME280_Library change <Adafruit_BME280.h> --> 29 lines comment #include <Adafruit_Sensor.h>, BME280 this library does not use this header file, so comment out and no problem

collote69, 2019-02-15:
e commented #include <Adafruit_Sensor.h> but it gives me these errors I do not know what I do wrong

esp32-camera-bme280:56:1: error: 'Adafruit_BME280' does not name a type

 Adafruit_BME280 bme;

 ^

C:\Users\Ivan\Pictures\esp32\esp32-camera-bme280\esp32-camera-bme280.ino: In function 'void drawFrame2(OLEDDisplay*, OLEDDisplayUiState*, int16_t, int16_t)':

esp32-camera-bme280:92:33: error: 'bme' was not declared in this scope

         temp = "Temp:" + String(bme.readTemperature()) + " *C";

                                 ^

C:\Users\home\Pictures\esp32\esp32-camera-bme280\esp32-camera-bme280.ino: In function 'void setup()':

esp32-camera-bme280:135:10: error: 'bme' was not declared in this scope

     if (!bme.begin(BEM280_ADDRESS)) {

          ^

Se encontraron varias bibliotecas para "WiFi.h"
Usado: C:\Users\home\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\WiFi
 No usado: C:\Users\Ivan\Pictures\esp32\arduino-1.8.8\libraries\WiFi
Se encontraron varias bibliotecas para "Wire.h"
Usado: C:\Users\home\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\Wire
 No usado: C:\Users\home\Pictures\esp32\arduino-1.8.8\libraries\Wire
exit status 1
'Adafruit_BME280' does not name a type

lewisxhe, 2019-02-15:
> e commented #include <Adafruit_BME280.h> but it gives me these errors I do not know what I do wrong
> 
> esp32-camera-bme280:56:1: error: 'Adafruit_BME280' does not name a type
> 
> Adafruit_BME280 bme;
> 
> ^
> 
> C:\Users\Ivan\Pictures\esp32\esp32-camera-bme280\esp32-camera-bme280.ino: In function 'void drawFrame2(OLEDDisplay*, OLEDDisplayUiState*, int16_t, int16_t)':
> 
> esp32-camera-bme280:92:33: error: 'bme' was not declared in this scope
> 
> ```
>      temp = "Temp:" + String(bme.readTemperature()) + " *C";
> 
>                              ^
> ```
> C:\Users\home\Pictures\esp32\esp32-camera-bme280\esp32-camera-bme280.ino: In function 'void setup()':
> 
> esp32-camera-bme280:135:10: error: 'bme' was not declared in this scope
> 
> ```
>  if (!bme.begin(BEM280_ADDRESS)) {
> 
>       ^
> ```
> Se encontraron varias bibliotecas para "WiFi.h"
> Usado: C:\Users\home\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\WiFi
> No usado: C:\Users\Ivan\Pictures\esp32\arduino-1.8.8\libraries\WiFi
> Se encontraron varias bibliotecas para "Wire.h"
> Usado: C:\Users\home\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\Wire
> No usado: C:\Users\home\Pictures\esp32\arduino-1.8.8\libraries\Wire
> exit status 1
> 'Adafruit_BME280' does not name a type

No, not the #include <Adafruit_BME280.h> line, enter the `Adafruit_BME280_Library` library, usually in your `C:\Users\YourName\Documents\Arduino\libraries\Adafruit_BMP280_Library`, comment `Adafruit_BME280.h` Line 29 in the file

collote69, 2019-02-15:
hello sorry in the file H or CPP and commented on CPP and I get this.

float Adafruit_BME280::seaLevelForAltitude(float altitude, float atmospheric)

       ^

Se encontraron varias bibliotecas para "WiFi.h"
Usado: C:\Users\Ivan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\WiFi
 No usado: C:\Users\Ivan\Pictures\esp32\arduino-1.8.8\libraries\WiFi
Se encontraron varias bibliotecas para "Wire.h"
Usado: C:\Users\Ivan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\Wire
 No usado: C:\Users\Ivan\Pictures\esp32\arduino-1.8.8\libraries\Wire
Se encontraron varias bibliotecas para "SPI.h"
Usado: C:\Users\Ivan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\SPI
 No usado: C:\Users\Ivan\Pictures\esp32\arduino-1.8.8\libraries\SPI
exit status 1
Error compilando para la tarjeta ESP32 Wrover Module.

I'm sorry I do not know much about this

lewisxhe, 2019-02-15:
@collote69 Please replace your existing deletion with me. [Adafruit_BME280_Library](https://drive.google.com/open?id=1g0507PU4t9Dpp8fwfJCmdljwpknkb5VJ)

collote69, 2019-02-15:
Several libraries were found for "Wire.h"
Usado: C:\Users\home\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\Wire
 No usado: C:\Users\home\Pictures\esp32\arduino-1.8.7\libraries\Wire
Se encontraron múltiples librerías para "Adafruit_BME280.h"
Usado: C:\Users\home\Documents\Arduino\libraries\Adafruit_BME280_Library
 No usado: C:\Users\home\Documents\Arduino\libraries\BME280-I2C-ESP32-master
Several libraries were found for "SPI.h"
Usado: C:\Users\home\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\SPI
 No usado: C:\Users\home\Pictures\esp32\arduino-1.8.7\libraries\SPI
Several libraries were found for "WiFi.h"
Usado: C:\Users\home\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\WiFi
 No usado: C:\Users\home\Pictures\esp32\arduino-1.8.7\libraries\WiFi
exit status 1
Error compiling for the card ESP32 Wrover Module.

 and replaced what you have told me and this above is the error

collote69, 2019-02-15:
hello pardon and tried now canbiando with an ubuntu and arduino 1.8.8 and canbiando the bookstore by which you have passed me and this is the error:

 sketch/esp32-camera-bme280.ino.cpp.o:(.literal._Z5setupv+0x60): undefined reference to `startCameraServer()'
sketch/esp32-camera-bme280.ino.cpp.o: In function `setup()':
/home/comodo/Vídeos/esp32-camera-bme280/esp32-camera-bme280/esp32-camera-bme280.ino:215: undefined reference to `startCameraServer()'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for the card ESP32 Wrover Module.

collote69, 2019-02-15:
Sorry I just did what you tell me I'm replacing the libraries and I'm testing with arduino in ubuntu now the error is:

n file included from /home/comodo/.arduino15/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp32-camera/esp_camera.h:70:0
                 from /home/comodo/Images/esp32-camera-bme280/esp32-camera-bme280.ino:6:
/home/comodo/.arduino15/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp32-camera/sensor.h:83:24: error: conflicting declaration 'typedef struct _sensor sensor_t'
 typedef struct _sensor sensor_t;
                        ^
In file included from /home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.h:29:06,
                 from /home/comodo/Images/esp32-camera-bme280/esp32-camera-bme280.ino:5:
/home/comodo/Documentos/Arduino/libraries/Adafruit_Unified_Sensor/Adafruit_Sensor.h:137:3: note: previous declaration as 'typedef struct sensor_t sensor_t'
 } sensor_t;
   ^
In file included from /home/comodo/.arduino15/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp32-camera/esp_camera.h:70:0
                 from /home/comodo/Images/esp32-camera-bme280/esp32-camera-bme280.ino:6:
/home/comodo/.arduino15/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp32-camera/sensor.h:122:3: error: conflicting declaration 'typedef struct _sensor sensor_t'
 } sensor_t;
   ^
In file included from /home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.h:29:06,
                 from /home/comodo/Images/esp32-camera-bme280/esp32-camera-bme280.ino:5:
/home/comodo/Documentos/Arduino/libraries/Adafruit_Unified_Sensor/Adafruit_Sensor.h:137:3: note: previous declaration as 'typedef struct sensor_t sensor_t'
 } sensor_t;
   ^
/home/comodo/Images/esp32-camera-bme280/esp32-camera-bme280.ino: In function 'void button1Func ()':
esp32-camera-bme280: 67: 8: error: 'struct sensor_t' has no member named 'set_vflip'
     s-> set_vflip (s, in);
        ^
/home/comodo/Images/esp32-camera-bme280/esp32-camera-bme280.ino: In function 'void setup ()':
esp32-camera-bme280: 181: 8: error: 'struct sensor_t' has no member named 'set_framesize'
     s-> set_framesize (s, FRAMESIZE_QVGA);
        ^
Several libraries were found for "SPI.h"
Used: /home/comodo/.arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/SPI
 Not used: / home / comodo / arduino / libraries / SPI
exit status 1
'struct sensor_t' has no member named 'set_vflip'

lewisxhe, 2019-02-15:
@collote69 
Have you replaced it with the library I provided?

I see the error message or the previous error. According to your error message, this library location should be at `/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.h`

Jump to line 29, as shown below, close the header file, and then recompile.

[image]

collote69, 2019-02-15:
if I replaced the library that I had provided and I just commented on line 29 of how the image shows me and now this comes out:

/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:40:1: error: 'Adafruit_BME280' does not name a type
 Adafruit_BME280::Adafruit_BME280()
 ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:50:1: error: 'Adafruit_BME280' does not name a type
 Adafruit_BME280::Adafruit_BME280(int8_t cspin)
 ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:63:1: error: 'Adafruit_BME280' does not name a type
 Adafruit_BME280::Adafruit_BME280(int8_t cspin, int8_t mosipin, int8_t misopin, int8_t sckpin)
 ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:75:6: error: 'Adafruit_BME280' has not been declared
 bool Adafruit_BME280::begin(TwoWire *theWire)
      ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp: In function 'bool begin(TwoWire*)':
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:77:2: error: '_wire' was not declared in this scope
  _wire = theWire;
  ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:78:2: error: '_i2caddr' was not declared in this scope
  _i2caddr = BME280_ADDRESS;
  ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:78:13: error: 'BME280_ADDRESS' was not declared in this scope
  _i2caddr = BME280_ADDRESS;
             ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:79:14: error: void value not ignored as it ought to be
  return init();
              ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp: At global scope:
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:89:6: error: 'Adafruit_BME280' has not been declared
 bool Adafruit_BME280::begin(uint8_t addr)
      ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp: In function 'bool begin(uint8_t)':
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:91:2: error: '_i2caddr' was not declared in this scope
  _i2caddr = addr;
  ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:92:2: error: '_wire' was not declared in this scope
  _wire = &Wire;
  ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:93:14: error: void value not ignored as it ought to be
  return init();
              ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp: At global scope:
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:104:6: error: 'Adafruit_BME280' has not been declared
 bool Adafruit_BME280::begin(uint8_t addr, TwoWire *theWire)
      ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp: In function 'bool begin(uint8_t, TwoWire*)':
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:106:5: error: '_i2caddr' was not declared in this scope
     _i2caddr = addr;
     ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:107:2: error: '_wire' was not declared in this scope
  _wire = theWire;
  ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:108:14: error: void value not ignored as it ought to be
  return init();
              ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp: At global scope:
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:117:6: error: 'Adafruit_BME280' has not been declared
 bool Adafruit_BME280::begin(void)
      ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp: In function 'bool begin()':
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:119:5: error: '_i2caddr' was not declared in this scope
     _i2caddr = BME280_ADDRESS;
     ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:119:16: error: 'BME280_ADDRESS' was not declared in this scope
     _i2caddr = BME280_ADDRESS;
                ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:120:2: error: '_wire' was not declared in this scope
  _wire = &Wire;
  ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:121:14: error: void value not ignored as it ought to be
[... 296 lines cut ...]
       ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp: In function 'float readHumidity()':
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:548:28: error: 'BME280_REGISTER_HUMIDDATA' was not declared in this scope
     int32_t adc_H = read16(BME280_REGISTER_HUMIDDATA);
                            ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:554:18: error: 't_fine' was not declared in this scope
     v_x1_u32r = (t_fine - ((int32_t)76800));
                  ^
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:556:48: error: '_bme280_calib' was not declared in this scope
[... cut ...]
/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp:604:7: error: 'Adafruit_BME280' has not been declared
 float Adafruit_BME280::seaLevelForAltitude(float altitude, float atmospheric)
       ^
Se encontraron varias bibliotecas para "SPI.h"
Usado: /home/comodo/.arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/SPI
 No usado: /home/comodo/arduino/libraries/SPI
exit status 1
Error compilando para la tarjeta ESP32 Wrover Module.

yes this library location should be at `/home/comodo/Documentos/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.h`

collote69, 2019-02-15:
Hello, I did the last thing you said and uncomment #include "Adafruit_BME280.h" line 29 of the Adafruit_BME280.cpp library file and it works thanks so much and sorry for the inconvenience

closed 2019-02-15 by collote69