lewisxhe/esp32-camera-series #8: sensor_t struct error on compilation

sorriso93, 2019-04-02:
Hello below the errors.
What board should I use to compile it?
Tried TTGO Lora32-OLED v1 and also ESP32-WROOM

in file included from /Users/yoghibear/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp32-camera/esp_camera.h:70:0,
                 from /Users/yoghibear/Desktop/progr esp2866/18_camera_pir/esp32-camera-bme280/esp32-camera-bme280.ino:6:
/Users/yoghibear/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp32-camera/sensor.h:83:24: error: conflicting declaration 'typedef struct _sensor sensor_t'
 typedef struct _sensor sensor_t;
                        ^
In file included from /Users/yoghibear/Documents/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.h:29:0,
                 from /Users/yoghibear/Desktop/progr esp2866/18_camera_pir/esp32-camera-bme280/esp32-camera-bme280.ino:5:
/Users/yoghibear/Documents/Arduino/libraries/Adafruit_Sensor/Adafruit_Sensor.h:137:3: note: previous declaration as 'typedef struct sensor_t sensor_t'
 } sensor_t;
   ^
In file included from /Users/yoghibear/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp32-camera/esp_camera.h:70:0,
                 from /Users/yoghibear/Desktop/progr esp2866/18_camera_pir/esp32-camera-bme280/esp32-camera-bme280.ino:6:
/Users/yoghibear/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp32-camera/sensor.h:122:3: error: conflicting declaration 'typedef struct _sensor sensor_t'
 } sensor_t;
   ^
In file included from /Users/yoghibear/Documents/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.h:29:0,
                 from /Users/yoghibear/Desktop/progr esp2866/18_camera_pir/esp32-camera-bme280/esp32-camera-bme280.ino:5:
/Users/yoghibear/Documents/Arduino/libraries/Adafruit_Sensor/Adafruit_Sensor.h:137:3: note: previous declaration as 'typedef struct sensor_t sensor_t'
 } sensor_t;
   ^
/Users/yoghibear/Desktop/progr esp2866/18_camera_pir/esp32-camera-bme280/esp32-camera-bme280.ino: In function 'void button1Func()':
esp32-camera-bme280:67:8: error: 'struct sensor_t' has no member named 'set_vflip'
     s->set_vflip(s, en);
        ^
/Users/yoghibear/Desktop/progr esp2866/18_camera_pir/esp32-camera-bme280/esp32-camera-bme280.ino: In function 'void setup()':
esp32-camera-bme280:181:8: error: 'struct sensor_t' has no member named 'set_framesize'
     s->set_framesize(s, FRAMESIZE_QVGA);
        ^
Multiple libraries were found for "WiFi.h"
 Used: /Users/yoghibear/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/WiFi
 Not used: /Applications/Sviluppo/Arduino.app/Contents/Java/libraries/WiFi
Multiple libraries were found for "Adafruit_Sensor.h"
 Used: /Users/yoghibear/Documents/Arduino/libraries/Adafruit_Sensor
 Not used: /Users/yoghibear/Documents/Arduino/libraries/Adafruit_Unified_Sensor
exit status 1
'struct sensor_t' has no member named 'set_vflip'

sorriso93, 2019-04-02:
Sorry, didn't read about the comment out for include adafruit_sensor.h in bme library. now it's fine

closed 2019-04-02 by sorriso93

Patriboom, 2024-01-24:
You'll find the answer, I guess, hear:   https://forum.arduino.cc/t/conflicitng-declaration-in-adafruit_sensor-esp32-camera/586568/2