lewisxhe/esp32-camera-series #11: Problem with: undefined reference to `startCameraServer()

andresgs375, 2019-05-03:
Hello:
I'm having troubles with this esp32-camera-series program.
 ESP32 Wrover module selected, Huge APP (3MB No OTA) selected, #include <Adafruit_Sensor.h line commented. I am using the TTGO T-Camera ESP32 WROVER & PSRAM Camera Module ESP32-WROVER-B OV2640 Camera Module 0.96 OLED
The error is:
sketch\esp32-camera-series.ino.cpp.o: In function `setup()':

C:\Users\Andres\Documents\MiArduino\esp32-camera-series\esp32-camera-series/esp32-camera-series.ino:342: undefined reference to `startCameraServer()'
I know there was a past issue with the same problem but the solution was refered to esp32-camera-bme280 sketch.I hope you can help, thanks in advance.

andresgs375, 2019-05-05:
Problem solved, it was the same problem as the other issue, the sketch was moved out of the same folder and the arduino IDE was opening only one tab of the three.

voir64, 2019-06-09:
bonjour
quand je compile arduino me répond
Erreur de compilation pour la carte ESP32 Wrover Module
que puis je faire ?

uliatgithub, 2020-06-22:
@andresgs375 Can you please explain what exactly you have done to solve the problem, ploease?
I am using the ESP32-CAM board and get the same error. I hope withyour description I can understand first what exactly is wrong and then fix it.
My error message:
.pio\build\esp32cam\src\main.cpp.o:(.literal._Z5setupv+0x38): undefined reference to `startCameraServer()'
.pio\build\esp32cam\src\main.cpp.o: In function `setup()':
C:\Users\Uli\Documents\PlatformIO\Projects\ESP32_CAM_Test1/src/main.cpp:98: undefined reference to `startCameraServer()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32cam\firmware.elf] Error 1

uliatgithub, 2020-06-23:
I found the problem as well. The app_httpd.cpp must be in the same directory as the main.cpp file. 
In my case: C:\Users\Uli\Documents\PlatformIO\Projects\ESP32_CAM_Test1\src.

shariq-azim, 2020-07-30:
So i was facing the same issue. but not sure what  was meant above. so after several i realized what ^this person means that all the files mentioned in the path https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Camera/CameraWebServer needs too  be downloaded into the same library folder where your sketch is present

MuhammadAjwad, 2022-01-03:
Hi, i face an issue with the 'startcameraserver', i'm using the ESP32 CAM that linked with Blynk server with the message :
sketch\Door_Lock__V._3_.ino.cpp.o:(.literal._Z5setupv+0x64): undefined reference to `startCameraServer()'

MuhammadAjwad, 2022-01-03:
I try to closed all the IDE tabs and restart my IDE software again but it didn't work, are there any way to solve this issue?

paynterf, 2025-01-29:
I had this same problem recently with the 'CameraWebServer' example project (C:\Users\Frank\Documents\Arduino\Libraries\arduino-esp32-master\libraries\ESP32\examples\Camera\CameraWebServer).  The problem arises if the 'app_httpd.cpp' source file that contains the definition for 'startCameraServer()' and 'setupLedFlash(int pin)' isn't in the compile/link set for the project.  I use MS Visual Studio 2022 with Visual Micro for my Arduino projects, so for me the solution was to simply add app_httpd.cpp to the project (right click on project name in Solution Explorer ->Add -> Existing Item -> app_httpd.cpp).  After doing that everything compiled and ran fine