lewisxhe/esp32-camera-series #15: 'httpd_resp_sendstr' was not declared in this scope

Stylesoftware, 2019-08-31:
Tried on esp 1.01 and 1.02.
Installed the required libraries.
Tried older and newer Arduino IDE.
Selected board: ESP Wrover (3MB Not OTA)
 
Used: C:\Users\xx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\WiFi
Used: C:\Users\xx\Documents\Arduino\libraries\esp8266-oled-ssd1306

Error:
```
sketch\app_httpd.cpp: In function 'esp_err_t stream_hmi_handler(httpd_req_t*)':

app_httpd.cpp:217:119: error: 'httpd_resp_sendstr' was not declared in this scope

     res = httpd_resp_sendstr(req, "HTTP/1.1 200 OK\r\nContent-Type: multipart/x-mixed-replace; boundary=frame\r\n\r\n");

```

the-programmer, 2019-09-01:
Hi,

I managed to fix this error for my own code.
Just add the following snippet to the top of "app_httpd.cpp":
`static inline esp_err_t httpd_resp_sendstr(httpd_req_t *r, const char *str) {
    return httpd_resp_send(r, str, (str == NULL) ? 0 : strlen(str));
}`
The code is from https://github.com/espressif/esp-idf/blob/a7e8d87d3e5ccc9e5ffcd701a1bac587ba4f43ea/components/esp_http_server/include/esp_http_server.h#L1041
I guess it is dropped by espressif for some reason.

Stylesoftware, 2019-09-02:
Good stuff, cheers

lewisxhe, 2019-09-23:
already fixed @the-programmer Thank you for your answer

closed 2019-09-23 by lewisxhe