lewisxhe/esp32-camera-series #18: New Layout Rev.1.6.2 of TTGO T-Camera with Mic

oliverathome, 2020-03-12:
A new Version of the PCB is in the market, labled as OV2640_1.6.2.
[image]

This Version has no ESP32 WROVER B module. The ESP32 is directly soldered to the PCB.

Some IO's seem to have changed. 
The Button has changed from IO0 (V16 Board) to IO15 (V162 Board).
IO15 was connected to Camera Y8 (v16 Board). 

The TTGO_T_CAMERA_MIC_V16 Pinout definition does not work with this Board V162.

Is there any updated pinout for this board avaliable ?

iwater, 2020-03-24:
this is my pin defined for version 1.6.2
#define PWDN_GPIO_NUM       -1
#define RESET_GPIO_NUM      -1
#define XCLK_GPIO_NUM       4
#define SIOD_GPIO_NUM       18
#define SIOC_GPIO_NUM       23

#define Y9_GPIO_NUM         36
#define Y8_GPIO_NUM         37
#define Y7_GPIO_NUM         38
#define Y6_GPIO_NUM         39
#define Y5_GPIO_NUM         35
#define Y4_GPIO_NUM         14
#define Y3_GPIO_NUM         13
#define Y2_GPIO_NUM         34
#define VSYNC_GPIO_NUM      5
#define HREF_GPIO_NUM       27
#define PCLK_GPIO_NUM       25
#define AS312_PIN           19
#define BUTTON_1            15

#define I2C_SDA             21
#define I2C_SCL             22

oliverathome, 2020-03-24:
Thank you @iwater.
Your definition is working with Layout 1.6.2.

closed 2020-03-24 by oliverathome