lewisxhe/esp32-camera-series #20: platformio.ini suggests PIO, but it's Arduino IDE

ai-bits, 2020-05-05:
As I´m still struggling with coupla dozen things in Arduino and connected Android projects, it took me a loooong time to realize this is NOT a PlatformIO project -- despite the existence of platformio.ini in the sketch folder!
On the wrong track, with the VERY limited PIO knowledge, I converted the .ino to .cpp as suggested by the PIO IDE, but failed with the compiler.

Could you please
1.  prominently note in the README that this is an **Arduino IDE thing** (despite the platformio.ini file)
2.  maybe state how it can be done in PIO

Thanks
G.

lewisxhe, 2020-06-22:
Just create a new src folder, change the ino to main.cpp, and then place the .ini at the same level as the src directory

ai-bits, 2020-07-01:
Thanks for the explanation. Sorry it took me a while to see it.
Wouldn't it be worthwhile to add this to the Code > Readme.md?
G.