lewisxhe/esp32-camera-series #21: Error run sketch for T_Camera_MINI_VERSION, mismatch code and docs

orou74, 2020-06-26:
Hello,
I tried to run the code for T_Camera_MINI_VERSION but at startup i have the message :

 [E][camera.c:1049] camera_probe(): Detected camera not supported.
18:08:21.047 -> [E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

I found a difference in pins number between the file **select_pins.h** for the mini version where 
#define Y9_GPIO_NUM      39

and the documentation tells :
#define Y9_GPIO_NUM      19

I tried with both values but same error message at reboot after upload !

Do you have any ideas of what can I test ?
Thank you.
P.S : Any idea how to get the original firmware of the card because now it is bricked (sniff)

rafaelpascale, 2020-07-08:
Hello, 

The same in my Camera Mini, I Just changed the Pin out and apperars Error camera and Guru Meditation error.

I didnt find any docs in ttgo camera about an example

orou74, 2020-07-09:
Hello,
Here is the original source code that we have at delivery. It works for me.

[https://github.com/Xinyuan-LilyGO/TTGO_Camera_Mini?spm=a2g0s.imconversation.0.0.69073e5f0zi7ij](url)